stm32-rs/stm32l4xx-hal #13: What are APB1R1 and APB1R2?

twilco, 2018-09-21:
Hi there!  I'm writing [a HAL for STM32L433xx processors](https://github.com/twilco/stm32l433-hal) and using your project as a model.  Unsurprisingly, they're quite similar :).

In your [rcc.rs](https://github.com/MabezDev/stm32l432xx-hal/blob/6adbc4120b8947fa247acc57be03ffbca7f020b9/src/rcc.rs#L23) file, I noticed something called `APB1R1` and `APB1R2`.  What is this?  I found no mention of it in [the datasheet](https://www.st.com/resource/en/datasheet/stm32l432kc.pdf), and googling didn't yield much either.

MabezDev, 2018-09-21:
Hello!

When developing HAL's / drivers for these chips generally the info you need is in the [reference manual](https://www.st.com/resource/en/reference_manual/dm00151940.pdf). Navigating to the RCC section in the reference manual you will find references to the APB1R1 & APB1R2 registers.

Hope this helps :)

twilco, 2018-09-21:
Awesome, thank you!  I'm new to the ARM ecosystem coming from AVRs, where everything can be found in great detail just in the datasheets.  Suffice to say, these seem to be quite a bit more complex 😅.  It's good to know these reference manuals exist!

Would you mind if I e-mailed you with any further questions that come up?  I don't anticipate many since this manual seems very thorough, but still.

MabezDev, 2018-09-22:
Sure! If you haven't already checkout the [Discovery book](https://rust-embedded.github.io/discovery/).

closed 2018-09-22 by MabezDev