stm32-rs/stm32l4xx-hal #24: I2C and SSD1306

nickray, 2018-12-16:
So great that you're working on the I2C, thanks! I'm trying to get this to work together with @jamwaffles's `ssd1306` crate, but... no luck so far. Seeing as you have your own OLED driver (is I2C working with your ssd1351?), I was wondering if you see an obvious issue in https://github.com/nickray/ssd1306/blob/2be6dda92c9521a5babd6154f0012d8679898704/examples/stm32l4_graphics_i2c_128x32.rs? All I know is I'm getting a `NACKF` upon `disp.init`. The display (I have two) is attached as GND/VDD/SCL/SDA --> GND/3V3/D1/D0.

MabezDev, 2018-12-16:
My ssd1351 display is spi, but I am using i2c for the [max17048 lipo fuel gauge](https://github.com/MabezDev/max17048) and that works great. 

>  I was wondering if you see an obvious issue in https://github.com/nickray/ssd1306/blob/2be6dda92c9521a5babd6154f0012d8679898704/examples/stm32l4_graphics_i2c_128x32.rs

Looks fine to me. Try and see if you get anything when writing the address out, if you get a nackf when writing the address that indicates the address is wrong(these displays sometimes have more than one address, check the back of the display) or the wiring is wrong. If you have a scope/logic analyzer, that will help a lot in debugging your issue.

nickray, 2018-12-17:
Thanks! The address tip did the trick - this particular display ended up listening on address `0x78`.

closed 2018-12-17 by nickray