stm32-rs/stm32l4xx-hal #29: Feature gate devices

MabezDev, 2019-01-07:
Currently all the features are effectively useless as they are not device specific.

Things to consider:

- Support specific chips or support chip families instead
- L4+ family of chips?

Currently supporting chip families, but it may be better to go the route of f4 hal by using specific chips.

twilco, 2019-01-25:
Are we limited by the fact that the [stm32-rs library only specifies families](https://github.com/stm32-rs/stm32-rs/blob/master/stm32l4/src/lib.rs) for the `l4` series whereas the `f4` series [has individual chips exposed](https://github.com/stm32-rs/stm32-rs/blob/master/stm32f4/src/lib.rs)?

MabezDev, 2019-01-25:
I'm pretty sure those are the chip families for the f4, as in the f4 hal crate there are more devices than there are listed in in the stm32-rs crate. I've updated the issue to include a list of the devices, which should make it easier to track what needs feature gating.

twilco, 2019-01-26:
Thanks for clarifying.  I have an STM32L433 and would love to lend a hand with this issue.

mathk, 2019-03-25:
The L4x6 family is even split into those that have HSI16 and those that does not. How do you see that ?

ryan-summers, 2020-03-07:
I'm in the process of bringing up a project using the STM32L452 and I'm starting to see the need here (e.g. SPI2 is not exposed via the HAL, but the device has SPI1, SPI2, and SPI3). Since I'll likely need to implement this for my project, I'd be happy to contribute it back to this repository.

Of interest is Section 1.5 of [DM00151940](https://www.st.com/resource/en/reference_manual/dm00151940.pdf), which lists Product-specific features for the STM32L4 family:

[image]
[image]

mehmetalianil, 2020-08-20:
I have an STM32L475, and would love to contribute.

DrTobe, 2020-09-11:
As far as I have understood, there are no chip/device-specific feature gates implemented currently, since I do not see any appropriate features (e.g. `stm32l431`) in `Cargo.toml`, is that correct?

If so, what is the current plan to progress here? Feature-gating a whole device with all peripherals seems like a huge amount of work to me. Maybe, this is the reason why this has not yet been done? Would it be feasible to keep the device-families as features (each feature activating the minimum subset) but additionally introduce chip-specific features which depend on the device-family-features? With that approach, chip-specific peripherals could be added without breaking anything because the current minimum subset would be kept and automatically be activated when enabling one of the chip-specific features.

korken89, 2020-12-17:
Hi, this is correct.
I am not sure on how to proceed here, but we do need to feature gate on each MCU.
If you have an idea I'm very open to input!

mehmetalianil, 2021-01-19:
> If you fancy feature gating a device, you will need to go through reference manual for the chip family and feature gate the relevant peripherals. These are normally found in footnote e.g *not present on the stm32l4blahblah etc.

@MabezDev Footnotes in the Reference manual, or the tail of the source code? I would like to use feature gating to get accustomed to embedded-rust. Where shall I start?

syelman, 2021-02-09:
Hi I have  stm32l496zg and in a month or so I might able to use stm32l476 nucleo. I would like to help.

Note: I tried ADC on both of them and I had enable pwr for some pins for it to work. I also tried it for stm32l476 (with nucleo) and had to change some ascr registers. There are in my fork probably.

robbym, 2021-04-14:
Will this HAL crate support the STM32L4+ series? Like STM32L4R9VG?

MathiasKoch, 2021-04-14:
@robbym I don't see why not, but i guess it boils down to someone wanting to contribute the required PR. :+1:

Crzyrndm, 2021-10-14:
This is currently blocking me (see the ADC issue) leading to maintaining a fork for my current project. I don't want to continue this long term so want to at least make a start on this.

Is there any opinion on whether feature groups should continue to make feature checks less verbose? Potentially named as something obviously for internal use since many features are common to a large set of devices and AFAIK Rust doesn't have a good way to group features inside the source.
```
internal_stm32l4x6_line = []
...
internal_stm32l47x_variant = []
...
stm32l476 = [ "internal_stm32l47x_variant", "internal_stm32l4x6_line" ]
```
I'm on the fence because while I like the idea, the STML4 HAL lists them explicitly and so cross verification would be somewhat easier

Crzyrndm, 2021-12-11:
With the merge of #266, advancing this should be a bit easier. My focus is likely to be geared toward the L476 and L496 as I have hardware which would benefit, but other bits and pieces are likely (e.g. #264 )

A few comments
* #[cfg] and features are to me a relatively irritating part of Rust in that many bits of it feel incomplete (e.g. no ("internal" features)[https://github.com/rust-lang/cargo/issues/7769], no `#if ... #else ...`) which leads to a lack of clarity by requiring the top level features to be used everywhere. I'm guessing macros could help here (based on (cfg_if)[https://docs.rs/cfg-if/latest/cfg_if/] crate) but its really not an area of Rust that I know enough about so that's mostly speculation. Having a way of referring to L4x2 that didn't require listing the same 6 devices every time would be quite handy
* missing / incorrect PAC definitions definitely cause a few hiccups. I feel like this could benefit from a tracking issue,i.e. a list of known PAC issues with link, notes on any workaround (I've seem the wrong bit used a few times so it "mostly" works), and really just a place to see anything that is no longer blocked
* Finally, is there a reason to keep the separation of the USB_OTG feature? #266 didn't remove it or make it derived because I wasn't sure if there was more to it than just needing more specific feature gates

korken89, 2021-12-13:
For now, I think we are stuck in  `cfg`-hell as you say. Sadly I have no good idea on how to generalize this better. :(

> Having a way of referring to L4x2 that didn't require listing the same 6 devices every time would be quite handy

This would be great for not making copy-paste mistakes, but I don't think this exists today.

benjcollins, 2023-07-10:
Hi, I have a stm32l496 and seem to be missing PWM implementations for a number of the timers would feature gating this device allow me to use those features and if so how would I go about feature gating this chip?

rrwalton, 2023-12-28:
I have an stm32l475. Would definitely contribute if still needed.