stm32-rs/stm32l4xx-hal #39: Blinky.rs for stm32l475 LED PA5

tasosxak, 2019-02-25:
I modified the blinky.rs example to run for stm32l475:

```
//! Blinks an LED

#![deny(unsafe_code)]
// #![deny(warnings)]
#![no_std]
#![no_main]

extern crate cortex_m;
#[macro_use]
extern crate cortex_m_rt as rt;
extern crate cortex_m_semihosting as sh;
extern crate panic_semihosting;
extern crate stm32l4xx_hal as hal;
// #[macro_use(block)]
// extern crate nb;

use crate::hal::prelude::*;
use crate::hal::delay::Delay;
use crate::rt::ExceptionFrame;
use crate::rt::entry;

use core::fmt::Write;
use crate::sh::hio;

#[entry]
fn main() -> ! {

    let mut hstdout = hio::hstdout().unwrap();

    writeln!(hstdout, "Hello, world!").unwrap();

    let cp = cortex_m::Peripherals::take().unwrap();
    let dp = hal::stm32::Peripherals::take().unwrap();

    let mut flash = dp.FLASH.constrain(); // .constrain();
    let mut rcc = dp.RCC.constrain();

    // Try a different clock configuration
    let clocks = rcc.cfgr.hclk(8.mhz()).freeze(&mut flash.acr);
    // let clocks = rcc.cfgr
    //     .sysclk(64.mhz())
    //     .pclk1(32.mhz())
    //     .freeze(&mut flash.acr);

    // let mut gpioc = dp.GPIOC.split(&mut rcc.ahb2);
    // let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.afrh);

    let mut gpioa = dp.GPIOA.split(&mut rcc.ahb2);
    let mut led = gpioa.pa5.into_push_pull_output(&mut gpioa.moder, &mut gpioa.otyper);

    let mut timer = Delay::new(cp.SYST, clocks);
    loop {
        // block!(timer.wait()).unwrap();
        //timer.delay_ms(1000 as u32);
        led.set_high();
        // block!(timer.wait()).unwrap();
        //timer.delay_ms(1000 as u32);
        led.set_low();
    }
}

#[exception]
fn HardFault(ef: &ExceptionFrame) -> ! {
    panic!("{:#?}", ef);
}
```

```
cargo run  --release --example blinky --features="stm32l4x5,rt"
Compiling stm32l4xx-hal v0.3.6 (/home/tasosxak/Documents/RustProjects/stm32l4xx-hal)
```

```
Finished release [optimized + debuginfo] target(s) in 1.43s
     Running `gdb-multiarch -q -x openocd.gdb target/thumbv7em-none-eabi/release/examples/blinky`
Reading symbols from target/thumbv7em-none-eabi/release/examples/blinky...done.
0x00000000 in ?? ()
Breakpoint 1 at 0x8000a1a: file /home/tasosxak/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.7/src/lib.rs, line 550.
Breakpoint 2 at 0x8002982: file examples/blinky.rs, line 63.
Breakpoint 3 at 0x800064c: file /home/tasosxak/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-semihosting-0.5.1/src/lib.rs, line 78.
semihosting is enabled
Loading section .vector_table, size 0x188 lma 0x8000000
Loading section .text, size 0x283c lma 0x8000188
Loading section .rodata, size 0x1950 lma 0x80029e0
Start address 0x80007ca, load size 17172
Transfer rate: 24 KB/sec, 5724 bytes/write.
(gdb) continue
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.
```
The program gets stuck at this point. 

Openocd : 

```
openocd -f interface/stlink-v2-1.cfg -f target/stm32l4x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : clock speed 480 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.240714
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Info : device id = 0x10076415
Info : flash size = 1024kbytes
undefined debug reason 7 - target needs reset
semihosting is enabled
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
adapter speed: 480 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080007ca msp: 0x20018000, semihosting
adapter speed: 4000 kHz
Info : Padding image section 0 with 28 bytes
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x20000050 msp: 0x20018000, semihosting
Warn : block write succeeded
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
adapter speed: 480 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080007ca msp: 0x20018000, semihosting
Hello, world!
```

MabezDev, 2019-02-25:
I guessing your using [this discovery board?](https://www.st.com/en/evaluation-tools/b-l475e-iot01a.html)

According to [this from stm](https://www.st.com/content/dam/technology-tour-2017/hands-on-training_stm32.pdf) `PB14` blinks the LED, try that.

tasosxak, 2019-02-25:
```
error[E0609]: no field `pb14` on type `hal::gpio::gpiob::Parts`
  --> examples/blinky.rs:49:25
   |
49 |     let mut led = gpioa.pb14.into_push_pull_output(&mut gpioa.moder, &mut gpioa.otyper);
   |                         ^^^^ help: a field with a similar name exists: `pb1`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0609`.
error: Could not compile `stm32l4xx-hal`
```

In docs https://docs.rs/stm32l4xx-hal/0.3.6/stm32l4xx_hal/gpio/gpiob/index.html

Doesn't exist the struct PB14

MabezDev, 2019-02-25:
I've just opened #40, which includes all the `gpiob` pins.

You can try it out by putting this in you `Cargo.toml`
```
[dependencies.stm32l4xx-hal]
git = "https://github.com/mabezdev/stm32l4xx-hal"
rev = "78b0c33783b64d2eb55368b3003782142cd2af0e"
features = ["rt", "stm32l4x2"]
```

also `pb14` exists on the `gpiob` block, not `gpioa`.

tasosxak, 2019-02-25:
```
tasosxak@tasosxak:~/Documents/RustProjects/stm32l4xx-hal$ cargo run --release --example blinky --features="stm32l4x5,rt"
   Compiling stm32l4 v0.6.0
error: symbol `DEVICE_PERIPHERALS` is already defined
    --> /home/tasosxak/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32l4-0.6.0/src/stm32l4x5/mod.rs:1795:1
     |
1795 | static mut DEVICE_PERIPHERALS: bool = false;
     | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

error: Could not compile `stm32l4`.

To learn more, run the command again with --verbose.
```

Cargo.toml

```

[package]
name = "stm32l4xx-hal"
version = "0.3.6"
authors = ["Scott Mabin <MabezDev@gmail.com>"]
description = "Hardware abstraction layer for the stm32l4xx chips"
keywords = ["no-std", "stm32l4xx", "stm32l4", "embedded", "embedded-hal"]
categories = [
    "embedded",
    "hardware-support",
    "no-std",
]
repository = "https://github.com/MabezDev/stm32l4xx-hal"
readme = "README.md"
license = "MIT OR Apache-2.0"
exclude = [
	".travis.yml",
	".gitignore",
	"docs/",
	"docs/*"
]
edition = "2018"

[dependencies]
cortex-m = "0.5.8"
nb = "0.1.1"
stm32l4 = "0.6.0"
as-slice = "0.1"

[dependencies.cast]
version  = "0.2.2"
default-features = false

[dependencies.void]
version = "1.0.2"
default-features = false

[dependencies.stable_deref_trait]
default-features = false
version = "1.1"

[dependencies.embedded-hal]
version = "0.2.2"
features = ["unproven"]

[package.metadata.docs.rs]
features = ["rt", "stm32l4x2"]

[features]
rt = ["stm32l4/rt"]
stm32l4x1 = ["stm32l4/stm32l4x1"]
stm32l4x2 = ["stm32l4/stm32l4x2"]
stm32l4x3 = ["stm32l4/stm32l4x3"]
stm32l4x5 = ["stm32l4/stm32l4x5"]
stm32l4x6 = ["stm32l4/stm32l4x6"]
unproven = ["embedded-hal/unproven"]

[dev-dependencies]
panic-halt = "0.2.0"
panic-semihosting = "0.5.0"
cortex-m-semihosting = "0.3.1"
cortex-m-rt = "0.6.6"

[profile.dev]
incremental = false
codegen-units = 1

[profile.release]
codegen-units = 1
debug = true
lto = true

[dependencies.stm32l4xx-hal]
git = "https://github.com/mabezdev/stm32l4xx-hal"
rev = "78b0c33783b64d2eb55368b3003782142cd2af0e"
features = ["rt", "stm32l4x2"]
```

MabezDev, 2019-02-25:
My bad forgot you were running examples. Just do a pull and checkout `gpiob-full-range`.

tasosxak, 2019-02-25:
Solved.
Thank you!

closed 2019-02-25 by tasosxak