stm32-rs/stm32l4xx-hal #41: Temperature values from HTS221 sensor using stm32l4xx-hal

tasosxak, 2019-02-25:
Hi,

How to get temperature values using HTMS221 sensor using this HAL?

tasosxak, 2019-02-25:
Program:
```
#![deny(unsafe_code)]
// #![deny(warnings)]
#![no_std]
#![no_main]

extern crate cortex_m;
extern crate hts221;

#[macro_use]
extern crate cortex_m_rt as rt;
extern crate cortex_m_semihosting as sh;
extern crate panic_semihosting;
extern crate stm32l4xx_hal as hal;
// #[macro_use(block)]
// extern crate nb;

use crate::hal::prelude::*;
use crate::hal::delay::Delay;
use crate::rt::ExceptionFrame;
use crate::rt::entry;
use crate::sh::hio;

#[entry]
fn main() -> ! {

    let mut hstdout = hio::hstdout().unwrap();
    let cp = cortex_m::Peripherals::take().unwrap();
    let dp = hal::stm32::Peripherals::take().unwrap();
    let mut flash = dp.FLASH.constrain(); // .constrain();
    let mut rcc = dp.RCC.constrain();
    let clocks = rcc.cfgr.hclk(8.mhz()).freeze(&mut flash.acr);
    let mut gpiob = dp.GPIOB.split(&mut rcc.ahb2);
    let scl = gpiob.pb8.into_af4(&mut gpiob.moder, &mut gpiob.afrh);
    let sda = gpiob.pb9.into_af4(&mut gpiob.moder, &mut gpiob.afrh);
    let i2c = hal::i2c::I2c::i2c1(
        dp.I2C1,
        (scl, sda),
        1.mhz(),
        rcc.cfgr.freeze(&mut dp.FLASH.constrain().acr),
&mut rcc.agb1);

    let mut hts221 = hts221::Builder::new(i2c)
        .with_avg_t(hts221::AvgT::Avg256)
        .with_avg_h(hts221::AvgH::Avg512)
        .build()
        .unwrap();
    let relative_humidity = hts221.humidity_x2();
let temp_c = hts221.temperature_x8();

    
    let mut timer = Delay::new(cp.SYST, clocks);

    //writeln!(hstdout, "{}",temp_c.unwrap()).unwrap();

    loop {
       
    }
}

#[exception]
fn HardFault(ef: &ExceptionFrame) -> ! {
    panic!("{:#?}", ef);
}
```
Cargo command:
`cargo run --release --example temperature  --features="stm32l4x5,rt"`

Error:

```
error[E0277]: the trait bound `hal::gpio::gpiob::PB8<hal::gpio::Alternate<hal::gpio::AF4, hal::gpio::Input<hal::gpio::Floating>>>: hal::i2c::SclPin<hal::stm32::I2C1>` is not satisfied
  --> examples/temperature.rs:35:15
   |
35 |     let i2c = hal::i2c::I2c::i2c1(
   |               ^^^^^^^^^^^^^^^^^^^ the trait `hal::i2c::SclPin<hal::stm32::I2C1>` is not implemented for `hal::gpio::gpiob::PB8<hal::gpio::Alternate<hal::gpio::AF4, hal::gpio::Input<hal::gpio::Floating>>>`
   |
   = note: required by `<hal::i2c::I2c<hal::stm32::I2C1, (SCL, SDA)>>::i2c1`

error[E0277]: the trait bound `hal::gpio::gpiob::PB9<hal::gpio::Alternate<hal::gpio::AF4, hal::gpio::Input<hal::gpio::Floating>>>: hal::i2c::SdaPin<hal::stm32::I2C1>` is not satisfied
  --> examples/temperature.rs:35:15
   |
35 |     let i2c = hal::i2c::I2c::i2c1(
   |               ^^^^^^^^^^^^^^^^^^^ the trait `hal::i2c::SdaPin<hal::stm32::I2C1>` is not implemented for `hal::gpio::gpiob::PB9<hal::gpio::Alternate<hal::gpio::AF4, hal::gpio::Input<hal::gpio::Floating>>>`
   |
   = note: required by `<hal::i2c::I2c<hal::stm32::I2C1, (SCL, SDA)>>::i2c1`

error[E0609]: no field `agb1` on type `hal::rcc::Rcc`
  --> examples/temperature.rs:40:10
   |
40 | &mut rcc.agb1);
   |          ^^^^ help: a field with a similar name exists: `ahb1`

error: aborting due to 3 previous errors

Some errors occurred: E0277, E0609.
For more information about an error, try `rustc --explain E0277`.
error: Could not compile `stm32l4xx-hal`.

To learn more, run the command again with --verbose.
```

tasosxak, 2019-03-15:
```
#![deny(unsafe_code)]
#![no_main]
#![no_std]
extern crate stm32l4xx_hal;
extern crate hts221;

extern crate cortex_m;
#[macro_use(entry, exception)]
extern crate cortex_m_rt as rt;
#[macro_use(block)]
extern crate nb;
extern crate panic_semihosting;

use cortex_m_rt::entry;
use cortex_m_semihosting::{debug, hprintln};
use hts221::Builder;
use stm32l4xx_hal::rcc::RccExt;
use stm32l4xx_hal::gpio::GpioExt;
use stm32l4::stm32l4x5::interrupt;
use stm32l4xx_hal::i2c::SdaPin;
use stm32l4xx_hal::i2c::SclPin;
use stm32l4xx_hal::i2c::Error;
use stm32l4xx_hal::i2c::I2c;
use stm32l4xx_hal::stm32::I2C2;
use stm32l4xx_hal::flash::FlashExt;
use stm32l4xx_hal::time::U32Ext;
use stm32l4xx_hal::prelude::*;
use crate::stm32l4xx_hal::serial::Serial;

const PERIOD: u32 = 5_000_000;

#[entry]
fn main() -> ! {
    //hprintln!("Hello, world!").unwrap();

    
        let peripherals = stm32l4xx_hal::stm32l4::stm32l4x5::Peripherals::take().unwrap();
        let mut flash = peripherals.FLASH.constrain();
        let mut rcc = peripherals.RCC.constrain();
        let mut gpiob = peripherals.GPIOB.split(&mut rcc.ahb2);
        let clocks = rcc.cfgr.freeze(&mut flash.acr);
        let tx = gpiob.pb6.into_af7(&mut gpiob.moder, &mut gpiob.afrl);
        let rx = gpiob.pb7.into_af7(&mut gpiob.moder, &mut gpiob.afrl);

        let serial = Serial::usart1( peripherals.USART1, (tx, rx), 115_200.bps(), clocks, &mut rcc.apb2);
        let (mut tx, mut rx) = serial.split();
        let sent = b'A';
        block!(tx.write(sent)).ok();

        //let p = stm32l4xx_hal::stm32l4::stm32l4x5::Peripherals::take().unwrap();
block!(tx.write(b'h')).ok();
        let scl = gpiob.pb10.into_open_drain_output(&mut gpiob.moder, &mut gpiob.otyper).into_af4(&mut gpiob.moder, &mut gpiob.afrh);
        let sda = gpiob.pb11.into_open_drain_output(&mut gpiob.moder, &mut gpiob.otyper).into_af4(&mut gpiob.moder, &mut gpiob.afrh);
         block!(tx.write(b'k')).ok();
        let i2c = I2c::i2c2(
            peripherals.I2C2,
            (scl,sda),
            1.mhz(),
            rcc.cfgr.freeze(&mut flash.acr),
           &mut rcc.apb1r1
            );
         block!(tx.write(b'z')).ok();
       let mut hts = hts221::Builder::new(i2c)
        .with_avg_t(hts221::AvgT::Avg256)
        .with_avg_h(hts221::AvgH::Avg512)
        .build()
        .unwrap();
       
        let x  = 4;//hts.temperature_x8().unwrap() / 8;
        block!(tx.write(b'Y')).ok();
        block!(tx.write(x as u8)).ok();

      //hprintln!("temperature: {}",hts.temperature_x8().unwrap() / 8);

    loop {}
}
```

I get runtime error, after this command ` block!(tx.write(b'h')).ok();`

MabezDev, 2019-08-28:
Hi @tasosxak when you get to `rust_begin_unwind` if you continue again in gdb you will get the panic message which will help diagnose the issue.

closed 2019-09-02 by MabezDev