stm32-rs/stm32l4xx-hal #43: Pin alternate functions incorrect for chips that are not STM32L4x2

FrozenDroid, 2019-03-01:
I am trying to use I2C, but it won't work.
While trying to research why it won't work, I think I've found why. I am using the STM32L476VG, and if I'm correct @MabezDev is using the STM32L432KC?
If you compare https://www.st.com/resource/en/datasheet/stm32l476vg.pdf (page 92) and https://www.st.com/resource/en/datasheet/stm32l432kc.pdf (page 55). You will see that on the L476VG, PA9 and PA10 do not have I2C capabilities.

MabezDev, 2019-03-08:
You're right as of right now most of the development has been done with the l432 boards. There is a tracking issue #29 for other devices like yours. The plan is to go the same way as the f4 hal an feature gate each chip.

lucazulian, 2019-09-15:
I am trying to use I2C and I don't understand why. @FrozenDroid can you explain me the way to fix the problem?

FrozenDroid, 2019-09-15:
@lucazulian please explain your problem and I can try to help you.

lucazulian, 2019-09-15:
I'm trying to connect with a vl53l0x sensor via I2C but using a STM32L432KC board but I've got alway a `Nack` error on ports gpioa.pa9 and gpioa.pa10. The sensor works on arduino, and I thought the problem was related with this issue.

FrozenDroid, 2019-09-15:
No, it's probably not related. This issue is about chips that aren't the STM32L432KC. If you can give me some code, I can maybe help. One thing I found myself wasting time on was the difference between address byte sizes between AVR and ARM. Try shifting the address to the left one plane (addr << 1)

lucazulian, 2019-09-15:
Yeah, with your suggestion now it works....I can't explain me why. Anyway, thanks for the help

closed 2021-05-02 by korken89