stm32-rs/stm32l4xx-hal #59: Flash programming

nickray, 2019-05-28:
While I'm still iterating on the [Flash traits](https://github.com/nickray/stm32l4xx-hal/blob/wip/src/extra_traits/flash.rs), the [implementation](https://github.com/nickray/stm32l4xx-hal/blob/wip/src/flash.rs#L182-L279) is ready [for consumption](https://github.com/nickray/stm32l4xx-hal/blob/wip/examples/flash.rs#L36-L63).

My question is how we'd want to handle "more than unproven" traits, as there seems to be [no consensus and no progress](https://github.com/rust-embedded/embedded-hal/issues/28) in the case of Flash (and others...) currently. I certainly would like to retain the freedom to iterate the trait API while I start using it in earnest.

One tricky issue is the fact that in a general Flash trait, it's the size/alignment rules for reads, writes and erases are chip dependent. I'm currently using the `generic-array` approach, an alternative would be the "config crate" approach demo'd in https://github.com/jamesmunns/mcf-config-demo

Options are:
1. treat as `unproven`
1. treat as `extra-traits`
1. include traits in HAL implementation
1. use separate `flash-traits` crate
1. not merge at all into this repository

Currently I'm doing 2+3+5.

smilykoch, 2019-07-05:
I would love to use this in an application, but i am using stm32l4x5. Would you be interested in adding those to the feature gates? And maybe merging it into master behind the extra-traits gate? Or alternatively creating a separate crate, so we can actually use it together with the newest master code on the hal crate?

nickray, 2019-07-05:
I can try and put a PR together, have been using this with littlefs. Might take a bit as my fork diverged quite a lot.

smilykoch, 2019-07-10:
@nickray Sounds good! Let me know if i can help with anything!

smilykoch, 2019-07-11:
@nickray Also, do you have an example of using the trait implementations (FlashRead & WriteErase) as trait bounds in function or struct declaration? 

I cannot work out how to use the generic_array stuff :/

nickray, 2019-07-13:
Hmm, I think you're skipping a serialization step there? My traits are to be implemented on the [HAL Flash peripheral](https://github.com/nickray/stm32l4xx-hal/blob/wip/examples/flash.rs#L27). User code then [writes byte slices](https://github.com/nickray/stm32l4xx-hal/blob/wip/src/extra_traits/flash.rs#L119) to a given flash address. I don't think it makes sense to use them as trait bounds on data structs, certainly not in the HAL. [Serde](https://github.com/serde-rs/serde) has a bunch of formats with `no_std` implementations. 

The generic array stuff is just used because the granularity of reading/erasing/writing flash depends on the microcontroller. An alternative would be to use @jamesmunns' idea linked above.

smilykoch, 2019-07-15:
Ohh.. Perhaps i should have been more clear.

I want to make a library crate, with a bunch of functions that take in an interface implementing ```Read``` & ```WriteErase```.

So to use my example above, it would look something like:

```
impl<F> MyStruct<F>
where
  F: Read<?> + WriteErase<?,?>
{
  pub fn new(dev: F, pageLimits: ?) {}
}
```

And then it can be used as 
```
let mut flash = hal::flash::Flash::new(dp.FLASH);
let FlashManager = MyStruct::new(flash, ?);
```

Is this a bad/wrong use case? I am still fairly new in the rust world of trait bounds

nickray, 2019-07-15:
It's not "bad", it's just not the intended use case. The goal is to be able to read/write arrays of raw bytes. If you want to read/write structs, you'll have to implement it yourself.

MathiasKoch, 2020-04-01:
I think this will be fixed by #91 if i'm not mistaken?

closed 2020-09-16 by MabezDev