stm32-rs/stm32l4xx-hal #60: Question: Remaining serial interfaces

smilykoch, 2019-06-18:
Hi.

I am in a situation, where i need USART3 & 4 on my stm32l475. I have implemented these, and they are tested and working fully. Now i would like to add a PR on these, but was thinking i would add the full suite of uart and usart peripherals for the entire family. 

So the question is, if i want to contribute this, would you be interested in me doing it in the same fashion as https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/serial.rs, with regards to the feature gating? Or is there another way you want to go about it?

smilykoch, 2019-06-18:
Even better proposal:
Would you be into changing the serial interfaces to work in the same manner as implemented in the stm32l0xx-hal crate? https://github.com/stm32-rs/stm32l0xx-hal/blob/master/src/serial.rs

This would allow to setup the pins to suiting alternate functions within the serial abstraction as: 
```
 let tx_pin = gpioa.pa9;
 let rx_pin = gpioa.pa10;

 // Configure the serial peripheral.
 let serial = dp
     .USART2
     .usart((tx_pin, rx_pin), serial::Config::default(), &mut rcc)
     .unwrap();
```
I personally think this is a very clean way of doing this. But i think it might be reliant on #56 ?

MabezDev, 2019-08-28:
Closed by #65

closed 2019-08-28 by MabezDev