stm32-rs/stm32l4xx-hal #74: Use of four PWM

lucazulian, 2019-11-08:
From the example https://github.com/stm32-rs/stm32l4xx-hal/blob/master/examples/pwm.rs I'm trying to use the four pwn, but with this code only the pwm.3 works. Is it possible to use all the pwm exits? What is wrong with my implementation?

```
//! Testing PWM output

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate panic_halt;

// use cortex_m::asm;
use stm32l4xx_hal::{
    prelude::*,
    stm32,
    delay,
};
use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    let c = cortex_m::Peripherals::take().unwrap();
    let p = stm32::Peripherals::take().unwrap();

    let mut flash = p.FLASH.constrain();
    let mut rcc = p.RCC.constrain();

    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let mut gpioa = p.GPIOA.split(&mut rcc.ahb2);

    // TIM2
    let c1 = gpioa.pa0
        .into_push_pull_output(&mut gpioa.moder, &mut gpioa.otyper)
        .into_af1(&mut gpioa.moder, &mut gpioa.afrl);
    let c2 = gpioa.pa1
        .into_push_pull_output(&mut gpioa.moder, &mut gpioa.otyper)
        .into_af1(&mut gpioa.moder, &mut gpioa.afrl);
    let c3 = gpioa.pa2
        .into_push_pull_output(&mut gpioa.moder, &mut gpioa.otyper)
        .into_af1(&mut gpioa.moder, &mut gpioa.afrl);
    let c4 = gpioa.pa3
        .into_push_pull_output(&mut gpioa.moder, &mut gpioa.otyper)
        .into_af1(&mut gpioa.moder, &mut gpioa.afrl);

    let mut pwm = p
        .TIM2
        .pwm(
            (c1, c2, c3, c4),
            1.khz(),
            clocks,
            &mut rcc.apb1r1,
        );

    let mut pwma = pwm.0;
    let mut pwmb = pwm.1;
    let mut pwmc = pwm.2;
    let mut pwmd = pwm.3;
   
    let maxa = pwma.get_max_duty();
    let maxb = pwmb.get_max_duty();
    let maxc = pwmc.get_max_duty();
    let maxd = pwmd.get_max_duty();

    pwma.enable();
    pwmb.enable();
    pwmc.enable();
    pwmd.enable();

    let mut timer = delay::Delay::new(c.SYST, clocks);
    let second: u32 = 100;

    // NB: if the pins are LEDs, brightness is not
    //     linear in duty value.
    loop {
        pwma.set_duty(maxa);
        pwmb.set_duty(maxb);
        pwmc.set_duty(maxc);
        pwmd.set_duty(maxd);
        timer.delay_ms(second);
        // asm::bkpt();

        pwma.set_duty(maxa/11*10);
        pwmb.set_duty(maxb/11*10);
        pwmc.set_duty(maxc/11*10);
        pwmd.set_duty(maxd/11*10);
        timer.delay_ms(second);
    }
}
```

MabezDev, 2019-11-08:
I'm a little unfamiliar with the PWM interface. @nickray would you mind taking a look?

nickray, 2019-11-08:
So I only ever tested this with an RGB led, so likely missing implementation or implementation error (of the PWM). I'll have a look later, ping me if I forget.

lucazulian, 2019-11-08:
@nickray perfect, can I help you with something? Is there a way to use another timer instead of TIM2?

nickray, 2019-11-08:
Hey @lucazulian I don't use this chip anymore, so I think you'll have to fix this with a PR :) On first glance I can't see what's wrong.

I assume I cribbed this from another stm32-rs HAL, I think the f3/f4 are the most advanced ones, but can't find anything there. So I think I based the implementation here on https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/pwm.rs. There are some comments in there, I hope they help! It would probably make sense to:

1. figure out how the remapping works
2. implement things with a macro instead of case-by-case

lucazulian, 2019-11-09:
@nickray I've tried with only 2 pwm but only one seem to work

lucazulian, 2019-11-09:
I think the problem is related with
```unsafe { (*$TIMX::ptr()).ccer.write(|w| w.cc4e().set_bit()) }```
into Pwm<$TIMX, C4>. If i remove that line the next pwm works and so on

closed 2019-11-12 by lucazulian