stm32-rs/stm32l4xx-hal #76: MonoTimer alway 0 value

lucazulian, 2019-11-09:
I'm using a loop with `MonoTimer` but it seems not working. The part of the code I'm using:
```
#[entry]
fn main() -> ! {
        let pa = cortex_m::Peripherals::take().unwrap();
        let dwt = pa.DWT;

        let time = MonoTimer::new(dwt, clocks);

    loop {
    	// writes on serial interface
        writeln!(tx, "Hello time {}\r\n", time.now().elapsed()).unwrap();   
    }
}
```
But it can see always:
```
Hello time 0
Hello time 0
...
```
What am i doing wrong?

MabezDev, 2019-11-09:
If you are not using a debugger to run the program, you may [need to  enable tracing in the debug control block](https://github.com/MWatch/kernel/blob/master/src/main.rs#L114) else the DWT may not do anything.

lucazulian, 2019-11-09:
Ok, is there a way to have a `MonoTimer` without debugger? How do I fetch the "current time"?

nickray, 2019-11-09:
Take the CorePeripherals in `cortex_m` too, and call the two methods mabez linked on them?

nickray, 2019-11-09:
I'm surprised DWT is in Peripherals, so maybe DCB is there too? Also have a look at https://docs.rs/cortex-m/0.6.1/cortex_m/peripheral/struct.DCB.html. Unfortunately, due to docs.rs limitations, not all chip documentation is rendered there, but if you build stm32l4 and run `cargo doc --open` locally, that might help you find things.

nickray, 2019-11-10:
Did this work out?

closed 2019-11-19 by MabezDev