stm32-rs/stm32l4xx-hal #81: Serial error with USART1

lucazulian, 2019-11-26:
I'm using the example in `examples/serial.rs` but I'm not able to receive data, even if I can see data sent from the stm. What is wrong with that implementation?

MabezDev, 2019-11-26:
Going to need more information than that. What error are you getting? What pins and what board are you usung.?

lucazulian, 2019-11-26:
This is my implementation
```
// serial peripheral
        let tx = gpiob.pb6.into_af7(&mut gpiob.moder, &mut gpiob.afrl);
        let rx = gpiob.pb7.into_af7(&mut gpiob.moder, &mut gpiob.afrl);

        let mut serial = Serial::usart1(
            device.USART1,
            (tx, rx),
            Config::default().baudrate(USART1_BAUD_RATE.bps()),
            clocks,
            &mut rcc.apb2,
        );
```
I've got a rust unwild error

MabezDev, 2019-11-26:
Unwild error? What's that?

What board are you using?

lucazulian, 2019-11-26:
I'm using a STM32L432KC on pins D4 and D5 with openocd. When it tries to execute the send 
```
 let received = block!(rx.read()).unwrap();
```
it stops with that strange error

lucazulian, 2019-11-26:
The precise error is 
```
Breakpoint 3, rust_begin_unwind (info=0x200093f4) at src/main.rs:36
367         loop {}                                                
```

nickray, 2019-11-26:
I assume the unwrap panicked, can you hprintln! the error maybe?

lucazulian, 2019-11-26:
I've got differents error messages but I maybe it was my mistake. Thank you

closed 2019-11-26 by lucazulian