stm32-rs/stm32l4xx-hal #85: Serial Hardware flowcontrol

MathiasKoch, 2020-01-25:
I am currently working on a project where i need hardware flow control of the U(S)ART peripherals (CTS/RTS).
As far as i can see it would only be a matter of setting the two bits in `CR3`, and it should work fully, assuming the two gpio pins for CTS and RTS are already configured for the correct alternate function.

While implementing this, i see two paths and wanted to hear your take on which would be the correct way for this HAL.

## 1. option

```
pub enum FlowControl {
    FlowControlNone,
    FlowControlRts,
    FlowControlCts,
    FlowControlCtsRts,
}

pub struct Config {
    pub baudrate: Bps,
    pub parity: Parity,
    pub stopbits: StopBits,
    pub flow_control: FlowControl,
}

...
// Enable DMA transfers
usart.cr3.write(|w| w.dmat().set_bit().dmar().set_bit());

// Configure hardware flow control
match config.flow_control {
    FlowControl::FlowControlNone => usart.cr3.write(|w| w.rtse().clear_bit().ctse().clear_bit()),
    FlowControl::FlowControlRts => usart.cr3.write(|w| w.rtse().set_bit()),
    FlowControl::FlowControlCts => usart.cr3.write(|w| w.ctse().set_bit()),
    FlowControl::FlowControlCtsRts => usart.cr3.write(|w| w.rtse().set_bit().ctse().set_bit()),
};

// Enable One bit sampling method
usart.cr3.write(|w| w.onebit().set_bit());

// Configure baud rate
let brr = clocks.$pclkX().0 / config.baudrate.0;
assert!(brr >= 16, "impossible baud rate");
usart.brr.write(|w| unsafe { w.bits(brr) });
...
```

## 2. option

```
impl Pins<UART4> for ($TX, $RX, $CTS, $RTS) {
    const REMAP: u8 = 0;
    const FLOWCTL: u8 = 1;
}

impl Pins<UART4> for (PA0<Alternate<AF8, Input<Floating>>>, PA1<Alternate<AF8, Input<Floating>>>, PA15<Alternate<AF8, Input<Floating>>>, PB7<Alternate<AF8, Input<Floating>>>) {
    const REMAP: u8 = 0;
    const FLOWCTL: u8 = 1;
}
```

While keeping the already implemented `Pins<>` as
```
impl Pins<UART4> for (PA0<Alternate<AF8, Input<Floating>>>, PA1<Alternate<AF8, Input<Floating>>>) {
    const REMAP: u8 = 0;
    const FLOWCTL: u8 = 0;
}
```

The first option is definitely the faster one to write, but also opens up for configuring flowcontrol on the serial peripheral without putting the CTS/RTS pins into the correct AF mode.

I'd be happy to implement either or fully (all feature gates etc), i just don't wanna implement one of them, only to open a PR where the other one or maybe a third way is proposed.

MabezDev, 2020-01-25:
I have implemented the first approach on a different HAL, mainly because like you said it's faster. That said, I really like the second approach. 

I'll take either, my preference being the latter option.

MathiasKoch, 2020-01-25:
Cool!
I also have already implemented the first one, just so i could get on with my project locally, but just like you i like the latter approach best.

Mainly because it makes it harder for users to screw up configurations.

Ill take a swing at an implementation of the latter :+1:

PR #86 (2020-01-30): Implement hardware flow control for U(S)ART peripherals
Implement hardware flow control for U(S)ART peripherals, and implements remaining combinations of RX/TX pins.

The `pins!` macro can probably be optimized wrt. code lines, by using macro repetition.

Correct me if i'm wrong, but it also implements the remaining combinations of Rx/Tx pins, as i do believe they can be mixed and matches as one likes? eg. any USART1_RX with any USART1_TX pin, even though they are often used in the pairs they appear in on the chip pinout (PA2, PA3) etc.?

closes #85
closed 2020-01-30 by MabezDev