stm32-rs/stm32l4xx-hal #87: Countdown timers longer than 1 sec

MathiasKoch, 2020-01-28:
I think the current implementation of the Countdown trait for timers is decent at best, but I am having a hard time seeing how to use it for timers longer than 1 second (1 Hz)?

Maybe we should consider changing the Countdown implementation to use Microseconds as a time and then an `impl From<Hertz> for Microseconds`

as long as it includes the `From<Hertz>` it wouldn't even be a breaking change.

I think this can be used as a reference: https://github.com/jacobrosenthal/rppal/blob/46cfd55cc5d674d16901e697ef42fb1df9154dcd/src/hal.rs

Only thing i am unsure of is setting the prescalers correctly for all timers, but i can give it a shot, if people agree on changing the impl.

MathiasKoch, 2020-02-24:
Regarding this, would it be an option to use `core::time::Duration` as Time for timers? I think that would make MUCH more sense than using Hertz for a countdown timer?

MabezDev, 2020-03-03:
TIL that `core::time::Duration` exists...

Seems like using `core::time::Duration` pulls in 64 bit machinery which is a bit meh. I'm not that against it though. Alternatively, a `Microsecond` type backed by a u32 could work too.

> Only thing i am unsure of is setting the prescalers correctly for all timers

I haven't looked at the timer datasheets, but yeah there will be an upper limit on how many seconds the timer can actually count before wrapping around; whether we enforce this in the HAL is another question, perhaps a doc comment along the lines of "Don't exceed X seconds or stuff will break".

David-OConnor, 2021-01-05:
https://github.com/stm32-rs/stm32f3xx-hal/issues

davidji, 2021-08-09:
I'm just porting across from from stm32f1xx_hal, where timer.start(600.us()) works. Here's a link to the embedded hal documentation: https://docs.rs/embedded-hal/0.2.6/embedded_hal/timer/trait.CountDown.html. The example in the documentation won't compile. This is really confusing.