stm32-rs/stm32l4xx-hal #88: ADC implementation

FrozenDroid, 2020-02-10:
Hi,

For a project I'm working on, I need to use the ADC. Is work being done on it?
For now, I made a quick and crude implementation to use the ADC.
If no work is being done on it, I might take it upon me to handle this issue.

MabezDev, 2020-02-10:
I'm not aware of anyone working on it; I myself won't be adding any new features to this HAL, but I am more than happy to review PR's.

unizippro, 2020-02-24:
Hi @FrozenDroid
Are you still planning on taking on the ADC?
I would be happy to help or continue your work, if you have made any progress on it.

FrozenDroid, 2020-02-24:
@unizippro I am planning to continue and finish the implementation, but don't have the time right now.  
Feel free to continue on my work. 
https://github.com/FrozenDroid/stm32l4xx-hal/blob/master/src/adc.rs Is what I've started with.  
Let me know how you get on.

unizippro, 2020-02-25:
@FrozenDroid I have begun working on making a general implementation, still focused on OneShot. It is located on @MathiasKoch s fork in the adc branch, if you want to have a look.

unizippro, 2020-02-27:
Hey @FrozenDroid I have gotten quite far, but some of the stuff still needs to be generalized and put into macros. But i have hit a snag on my end. The ADC seems to be measuring to low.. It might be a problem with the hardware setup, but I have yet to find it. Could also just be a bug in the SW somewhere. Can be found [here](https://github.com/MathiasKoch/stm32l4xx-hal/tree/feature/adc).

FrozenDroid, 2020-02-27:
Is the measured value a constant value? And how low is too low? @unizippro Can you show me code you're using?

unizippro, 2020-03-02:
I'm running the ADC example i have just pushed. except it is running in a RTFM context.
It is not measuring a constant value, but it is a stable one. 
`RefVal: 1340, Vdda: 3723, TempVal: 739, Temp: -6392 C, Tcal: (104500, 131500)`
`RefVal: 1339, Vdda: 3725, TempVal: 737, Temp: -6422 C, Tcal: (104500, 131500)`
The degrees are in 1/100th. The Vref measured is too high, it should calculate to a Vdda of around. But the TempVal is to low, and should measure around 1050 to get a temp of around 28C.

unizippro, 2020-03-04:
After some work it seems to be a result of the clock setup, so fixing this should result in a correctly working ADC.

unizippro, 2020-03-06:
I have now corrected my clock setup, and compared it to an "identical" ADC setup using the C reference library. But they still get differing measurements. I have begun comparing the register setup after the ADC setup in the two code bases. But i have to put this aside until at least this Thursday.
If you have the time it would be really cool if you tested it, as it is working and should be producing relatively accurate results. 
Here is an example of how to use the lib 
``` Rust
hal::adc::adc_global_setup(
        hal::adc::CommonConfig::default()
        .clock_mode(hal::adc::ClockMode::HclkDiv4),
        &mut rcc.ahb2, &mut rcc.ccipr);
    let mut adc1 = Adc::adc1(
        dp.ADC1, hal::adc::Config::default(),
        &mut rcc.ahb2, &mut rcc.ccipr
    );

let mut t = VTemp::new();
t.enable();

let mut r = VRef::new();
r.enable();
cortex_m::asm::delay(40);
let temp_val: u16 =  match ctx.resources.adc1.read(&mut t){
    Ok(v) => v,
    Err(e)=> {
        info!("Err:{:?}", e);
        0
    }
};
let ref_val: u16 =  match adc1.read(&mut r){
    Ok(v) => v,
    Err(e)=> {
        info!("Err:{:?}", e);
        0
    }
};
let vdda = VRef::read_vdda(adc1);
let temp = VTemp::read(adc1);
let t_cal = VTemp::get_cal();
info!("RefVal: {:?}, Vdda: {:?}, TempVal: {:?}, Temp: {:?} C, Tcal: {:?}\r",
                ref_val, vdda, temp_val, temp, t_cal);
```

FrozenDroid, 2020-03-11:
Sorry for the lack of response, been fairly busy.
Unfortunately I don't have an STM32L4x5 so I won't be able to test it. 
I can maybe contribute to your code to make it work with the rest of the devices I have (4x2 and 4x6).

unizippro, 2020-03-12:
The datasheet covers both the 4x5 and 4x6, so it should be compatible with your 4x6 device. The feature gateing i set up is just so it'll run on my device. But it should be compatible with a lot more than just the 4x5.

unizippro, 2020-03-17:
I have to put this aside for now, as other deadlines are coming up. If you have the time to test that would be cool.
I have arrived at the point where the C hal and RS hal are quite close with the same clock setup. But some things are still quite off. 
C hal adc: ```Vrefval: 1506, Vref: 3312mV, Vtempval: 943, Vtemp: 29```
Rs hal adc: ```Vrefval: 1536, Vref: 3250mV, Vtempval: 658, Temp: -85.85```

SelmanOzleyen, 2020-09-25:
Hi @unizippro  and @FrozenDroid ;
I used some of the codes you discussed and created a fork with Adc Oneshot above and tested it with stm32l476 and stm32l496 using a potentiometer.

MathiasKoch, 2021-01-12:
I will close this for now, based on https://github.com/stm32-rs/stm32l4xx-hal/pull/173

closed 2021-01-12 by MathiasKoch