stm32-rs/stm32l4xx-hal #99: Replace mem::uninitialized() with mem::MaybeUninit

L0uisc, 2020-04-05:
I just compiled the crate for the first time, and I get a warning about a use of the deprecated `mem::uninitialized()` function on line 176 of `src/pwm.rs`. Since it is UB to call it, it would be a good idea to replace that asap with `mem::MaybeUninit`.

thalesfragoso, 2020-04-06:
The use of it in that case isn't UB, since the type is a zero size type. But we should definitely change that to `MaybeUninit`, since `mem::uninitialized()` is deprecated. Fortunately, the fix is very simple, just change that to `MaybeUninit::uninit().assume_init()`.

korken89, 2020-04-11:
To be more clear, and not potentially fall into problems in the future if the size changes, I'd use a `mem::transmute` instead of `MaybeUninit` there.

closed 2021-05-31 by korken89