stm32-rs/stm32l4xx-hal #107: implement Cancel trait for Timer

MarcoIeni, 2020-05-02:
Timer in [stm32f4xx-hal](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/timer.rs) crate implemented the [Cancel](https://docs.rs/crate/embedded-hal/0.2.3/source/src/timer.rs) trait.
You may be interested in doing the same for this crate.
Examples: 
- https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/timer-periph.rs
- https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/timer-syst.rs

korken89, 2021-05-31:
Hi, thanks for the request.
If you feel like helping and making a PR for this, it would be appreciated!

MarcoIeni, 2021-05-31:
sorry, I don't have the time for this, plus I don't have this microcontroller.

reitermarkus, 2022-05-03:
The `Cancel` trait is implemented now.

MarcoIeni, 2022-05-03:
Great, I will close this :)

closed 2022-05-03 by MarcoIeni