stm32-rs/stm32l4xx-hal #140: I2C timing for pclk1 != 8 Mhz

MathiasKoch, 2020-09-23:
Hi.

The current calculation for `i2c.timingr` seems to hold only in the case of using `pclk1` as clock source, and only if `pclk1 == 8 Mhz`.

In my case I would like to use `pclk1 = 80 Mhz`, which definitely does not work with the current implementation. 

I took a look at how ST calculates the `timingr` value (`presc`, `sdadel`, `scldel`, `sclh` & `scll`), and it looks like t hey basically "bruteforce" the value with the lowest error (Their java based implementation from STM32CubeMX can be found here: https://gist.github.com/MathiasKoch/1f070458054d1e3811d53b49be0b6f93)

I don't know how we intend to calculate this here? As i see it, we could do something similar, but it would NEVER be a zero-cost abstraction? Which might be fine, considering it is only during initialization?

MathiasKoch, 2020-09-23:
Hmm..
Going through it again, it might not be as off as i thought.. I made this to toy around with the implementation: https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=b8f20bbee9360f0cbf41d8efab3a7242

That said, I still don't think it is a very solid and easy to understand implementation

closed 2020-11-05 by MathiasKoch

lingsuls, 2024-03-29:
I used this code to calculate the official table【fI2CCLK=16KHz OR 8KHz】,but the output of this code is not matching.I want to know if this is normal.