stm32-rs/stm32l4xx-hal #144: <Output<OpenDrain>> InputPin trait

pdgilbert, 2020-10-03:
(Newbie, so don't assume I know what I'm talking about.)
Trying to work through a couple of examples using dht and dht11 crates I am getting errors

```
|     let r = dht11::Reading::read(&mut delay, &mut pin_a8);
    |                                              ^^^^^^^^^^^ the trait `embedded_hal::digital::v1::InputPin` is not implemented for `stm32l4xx_hal::gpio::gpioa::PA8<stm32l4xx_hal::gpio::Output<stm32l4xx_hal::gpio::OpenDrain>>`
    |
    = note: required because of the requirements on the impl of `embedded_hal::digital::v2::InputPin` for `stm32l4xx_hal::gpio::gpioa::PA8<stm32l4xx_hal::gpio::Output<stm32l4xx_hal::gpio::OpenDrain>>`
    = note: required because of the requirements on the impl of `dht_sensor::read::InputOutputPin<()>` for `stm32l4xx_hal::gpio::gpioa::PA8<stm32l4xx_hal::gpio::Output<stm32l4xx_hal::gpio::OpenDrain>>`
    = note: required by `dht_sensor::DhtReading::read`
```
I think this means that `Output<OpenDrain>>is not recognized as doing both output and input. I do know that this specification does build with several other HALs. (But note that I have not actual got it to work with any of them.) I see that `stm32f1xx-hal` has

```
impl InputPin for Generic<Output<OpenDrain>> {
...
```
which I do not see in `stm32l4xx-hal` Is this a bug or am I missing something?

korken89, 2021-05-31:
Hmm, seems like the GPIO module would need some extension to handle this.

DrTobe, 2021-10-02:
Shall we resolve this by also implementing the `InputPin` trait for pins configured as output pins? This may be irritating for push-pull pins but as far as I know the hardware can read back the actual pin level on an output pin. For open-drain pins, this _is_ meaningful, especially when setting the pin to high-impedance output state.

korken89, 2021-10-03:
To me or would make perfect sense to do so :)

pdgilbert, 2021-10-03:
(Related to #255)  I am wondering if recent changes in alternate functions may have fixed the open drain input problem. My current version of the example (further below) gives

```
   Compiling rust-integration-testing-of-examples v0.1.0 (/home/paul/githubClones/rust-integration-testing/dev-testing)
warning: unused imports: `AF0`, `Alternate`
  --> examples/zot.rs:45:43
   |
45 |     gpio::{gpioa::PA8, OpenDrain, Output, Alternate, AF0},
   |                                           ^^^^^^^^^  ^^^
   |
   = note: `#[warn(unused_imports)]` on by default

error[E0271]: type mismatch resolving `<PA8<Output<OpenDrain>> as stm32l4xx_hal::prelude::OutputPin>::Error == ()`
  --> examples/zot.rs:94:15
   |
94 |         match Reading::read(&mut delay, &mut dht_data) {
   |               ^^^^^^^^^^^^^ expected enum `Infallible`, found `()`
   |
   = note: required because of the requirements on the impl of `InputOutputPin<()>` for `PA8<Output<OpenDrain>>`
   = note: required by `dht_sensor::DhtReading::read`

error[E0277]: the trait bound `PA8<Output<OpenDrain>>: _embedded_hal_digital_InputPin` is not satisfied
  --> examples/zot.rs:94:41
   |
94 |         match Reading::read(&mut delay, &mut dht_data) {
   |                                         ^^^^^^^^^^^^^ the trait `_embedded_hal_digital_InputPin` is not implemented for `PA8<Output<OpenDrain>>`
   |
   = note: required because of the requirements on the impl of `stm32l4xx_hal::prelude::InputPin` for `PA8<Output<OpenDrain>>`
   = note: required because of the requirements on the impl of `InputOutputPin<()>` for `PA8<Output<OpenDrain>>`
   = note: required by `dht_sensor::DhtReading::read`

error: aborting due to 2 previous errors; 1 warning emitted
```
The example code is
<details>
<summary>Click to expand</summary>

```
//!  Measure the temperature and humidity from a DHT11 sensor and print with hprintln (to
//!  a gdb session).  The DHT11 data pin is connectted to pin A8 on the MCU board and has
//!  a pull up resistor. (18K ohms used in some testing.)
//!  The largest part of this file is the setup() functions used for each hal.
//!  These make the application code common.
//!
//!  Note that '--release' is needed when doing a run test on actual hardware. Otherwise
//!  code is too slow for the timeout set in the crate and run gives 'Error Timeout'.

#![deny(unsafe_code)]
#![no_main]
#![no_std]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;
use cortex_m_semihosting::hprintln;

//https://github.com/michaelbeaumont/dht-sensor
#[cfg(not(feature = "dht22"))]
use dht_sensor::dht11::Reading;
#[cfg(feature = "dht22")]
use dht_sensor::dht22::Reading;
use dht_sensor::*;

use embedded_hal::{
    blocking::delay::DelayMs,
};

// setup() does all  hal/MCU specific setup and returns generic hal device for use in main code.

// See dht-sensor git discussion in issues #1  and #2
//https://github.com/michaelbeaumont/dht-sensor/issues/1
//Regarding pulling the pin high to avoid confusing the sensor when initializing.
//Also more in comments in dht-sensor crate file src/lib.rs

#[cfg(feature = "stm32l4xx")]
use stm32l4xx_hal::{
    delay::Delay,
    gpio::{gpioa::PA8, OpenDrain, Output, Alternate, AF0},
    pac::{CorePeripherals, Peripherals},
    prelude::*,
};

#[cfg(feature = "stm32l4xx")]
use embedded_hal::digital::v2::OutputPin;

#[cfg(feature = "stm32l4xx")]
fn setup() -> (PA8<Output<OpenDrain>>, Delay) {
    let cp = CorePeripherals::take().unwrap();
    let p = Peripherals::take().unwrap();
    let mut flash = p.FLASH.constrain();
    let mut rcc = p.RCC.constrain();
    let mut pwr = p.PWR.constrain(&mut rcc.apb1r1);
    let clocks = rcc
        .cfgr
        .sysclk(80.mhz())
        .pclk1(80.mhz())
        .pclk2(80.mhz())
        .freeze(&mut flash.acr, &mut pwr);

    let gpioa = p.GPIOA.split(&mut rcc.ahb2);
    let mut pa8 = gpioa
        .pa8
        .into_open_drain_output(&mut gpioa.moder, &mut gpioa.otyper);
//        .into_af0_opendrain(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrh);

    // Pulling the pin high to avoid confusing the sensor when initializing.
    pa8.set_high().ok();

    // delay is used by `dht-sensor` to wait for signals
    let mut delay = Delay::new(cp.SYST, clocks); //SysTick: System Timer

    //  1 second delay (for DHT11 setup?) Wait on  sensor initialization?
    delay.delay_ms(1000_u16);

    (pa8, delay) //DHT data will be on A8
}

// End of hal/MCU specific setup. Following should be generic code.

#[entry]
fn main() -> ! {
    let (mut dht_data, mut delay) = setup(); //dht_data is usually pa8 in setup functions

    hprintln!("Reading sensor...").unwrap();

    loop {
        match Reading::read(&mut delay, &mut dht_data) {
            Ok(Reading {
                temperature,
                relative_humidity,
            }) => hprintln!("{} deg C, {}% RH", temperature, relative_humidity).unwrap(),
            Err(e) => hprintln!("Error {:?}", e).unwrap(),
        }

        // (Delay at least 500ms before re-polling, 1 second or more advised)
        // Delay 5 seconds
        delay.delay_ms(5000_u16);
    }
}

```
</details>

pdgilbert, 2021-11-05:
This was resolved by #263.

closed 2021-11-05 by pdgilbert