stm32-rs/stm32l4xx-hal #157: Time for a crates.io release?

cavokz, 2020-10-15:
I'd like to use the ExtiPin implementations for configuring interrupts on GPIO in a safe and ergonomic way. Do you have any plans for publishing a new release on crates.io? Can I help anyhow?

korken89, 2020-11-02:
I think it's time. @hannobraun @MathiasKoch @MabezDev, should we coordinate a realease after we get rest of the PRs in?

MabezDev, 2020-11-02:
Yes, sounds like a good idea.

MathiasKoch, 2020-11-03:
Sure thing!
Would love to get #148 in before if possible? I have time for a full review of that and other PR's tomorrow?

korken89, 2020-11-03:
Sounds good to me @MathiasKoch !

MathiasKoch, 2020-11-05:
#148 Is merged!

@MabezDev Will you make prepare for a new release at some point? Guess you are the only one with crates.io publish rights?

MathiasKoch, 2020-11-12:
Ping @MabezDev

MathiasKoch, 2020-12-10:
@MabezDev Could you make a new release to crates.io here?

MabezDev, 2020-12-11:
Apologies for the delay, I've opened https://github.com/stm32-rs/stm32l4xx-hal/pull/175 which is ready to go. Is there anything else we want to do before we release?

MathiasKoch, 2020-12-11:
Thank you! I think that should do it :+1: 

Sorry for pinging you continuously..

MabezDev, 2020-12-11:
> Sorry for pinging you continuously..

No worries, sorry for taking so long!

[v.6.0](https://crates.io/crates/stm32l4xx-hal) is now live!

closed 2020-12-11 by MabezDev