stm32-rs/stm32l4xx-hal #167: Maintainers wanted

MabezDev, 2020-10-28:
I am struggling to find the time and resources to review PR's at the moment. On top of which my personal use for the HAL has gone away for the time being, so motivation is at an all time low. I would hate to abandon this project, especially with the influx of new contributions but I simply cannot keep up. If you would like to help maintain this drop a comment in this thread :) 

cc: @korken89  @MathiasKoch  having been long time contributors, perhaps you would be interested in helping out?

korken89, 2020-10-28:
We use the HAL in our products so I can help with this!
I do not have so much time, but testing PRs and pulling them in should not be a problem :)

Thanks for all the hard work @MabezDev!

MabezDev, 2020-10-28:
Thank you @korken89 :) 

I've sent you and invitation.

hannobraun, 2020-10-29:
Thank you for all your work, @MabezDev!

I'm currently doing work for a client that uses this HAL, so I'll be active around here for the coming weeks, maybe months. Helping out with some reviews shouldn't be a problem.

I'm not sure if I should offer to help out as a maintainer though. I have no track record here (yet) and I'm unlikely to have a long-term interest in the HAL.

MathiasKoch, 2020-10-29:
First up, thanks your your amazing work @MabezDev !

We are running this particular chip and HAL in a production workload as well, and will keep doing so. 
I can help out with maintaining if @korken89 wants the help.

MabezDev, 2020-11-02:
Thanks guys, invites are sent :).

Appreciate the help!

closed 2020-11-02 by MabezDev

hannobraun, 2020-12-23:
Hi all. When I last wrote here, I expected that I'd be active in this HALs development for a while, but things turned out differently. First, I ended up doing much less work here than I initially expected. Second, my client project has now concluded, and no follow-up project is scheduled for the time being. This means I won't be very active here going forward, at least for now.

korken89, 2020-12-23:
Thanks for letting us know @hannobraun and thanks for the contributions!

David-OConnor, 2021-03-25:
I'm developing a multi-family HAL crate, and am using it to develop 2 commercial projects that use L4. I our philosophies diverge in a few areas, but we should collaborate.