stm32-rs/stm32l4xx-hal #180: ADC sometimes returns 0?

korken89, 2020-12-18:
Hi,

I am testing the newly merged ADC driver, and I pushed the update out on our system to give it a try.
I have the strange issue that some read the correct ADC value, and some always read 0 from the ADC (note: same firmware running on all).

Ping @hannobraun, have you seen something like this during your developments and what may have been the cause?
The only thing that I can think of is that the ADC calibration did not complete successfully. Could the 20us delay be too small due to clock errors (i.e. bad crystal or HSI derived clock)?
I have also checked the driver with the reference manual again, but to no avail. All seems correct.

hannobraun, 2020-12-18:
I don't think I've seen that, but I'm not sure.

I'm using the ADC as part of a [non-trivial test stand](https://github.com/braun-embedded/embedded-test-stand). I've seen the ADC value reported as 0 quite often, but that always happened while something else was going wrong. In particular, there's a second dev board involved in the testing, which sets a GPIO signal on the ADC pin. When the firmware on that board crashes, the signal is stuck at LOW, causing the ADC on the main dev board to always report 0.

So I don't think I've seen that issue. I've definitely never seen any weird ADC problems in isolation, without other failures that would explain them.

korken89, 2020-12-18:
Thanks, for the info!
Then I know that I need to start looking at the driver and usage more closely.
Adding delays in the driver seems to fix the problem, so it seems to be a timing bug in my case.

korken89, 2020-12-19:
It does not seem to be timing, but a know errata error in the STM32L412 series.

[image]

korken89, 2020-12-19:
I went through a bunch of STM32L4xx MCU's errata sheets and all state this error.
I think we need to redesign the ADC implementation to take care of this.

korken89, 2020-12-19:
I have been digging even more, what I have found is that whenever a system just get 0 from the ADC the ADC still reads 1 for `DEEPPWD`, indicating that the ADC is still in deep power down.
Not sure what causes this, and I cannot find any reference for this happening in the datasheet nor errata.

Any ideas on this @hannobraun ? I'm quite at a loss here.

PR #181 (2020-12-23): Fixed so the ADC driver is working on STM32L412 as well
I got the ADC to work on the STM32L412 as well now.
The main issue seems to have been that the enable of clocks was done after the reset of registers, which made the reset never take effect.
Moreover I did some checking of the internal clocks and added delays to compensate for bad clocks.
And finally I found that if one ran the measurements as fast as possible the disabling of the ADC had not finished until the next read, so now the `read` function waits for this.

Ping @hannobraun can you test if this still works on your HW?

Fixes #180
closed 2020-12-23 by MathiasKoch