stm32-rs/stm32l4xx-hal #188: Flash wait states in RCC

David-OConnor, 2021-01-10:
I'm going through the RCC module.

1: Check Reference manual section 3.3.3: There are 5 possible wait states. I think the current code is copy+pasted from a different HAL that has a different setup.

2: Wait state calculation is based on HCLK, not sysclk. Ie, you need to offset for the multiplier.

sourcebox, 2021-01-21:
I can confirm that the current wait states configuration is incorrect. Also, prefetch is also not enabled. Is this intentional?

FrozenDroid, 2021-03-23:
Is anyone working on this? This seems like a rather critical mistake. If not I'll try to whip up a proper solution.

FrozenDroid, 2021-03-25:
I believe this can be closed now :) @MathiasKoch

MathiasKoch, 2021-03-25:
Yes, i missed that it wasn't linked

closed 2021-03-25 by MathiasKoch