stm32-rs/stm32l4xx-hal #209: STM32L4R9 support

ButtNaked, 2021-04-12:
Greetings,

As follow up for stm32-rs/stm32-rs#455 
I want to help with L4R9 hal support.

What are the best first steps to accomplish this task?

Thx

MathiasKoch, 2021-04-14:
I think it is mostly a matter of adding the appropriate feature flag on all the peripherals and handling any edge cases or deviations that might come up.

The contribution would definitely be welcomed :+1:

ButtNaked, 2021-04-17:
Thx for answer
Okay, i' working on it, PR will follow soon™ 

@shymega maybe interested too

shymega, 2021-11-15:
Hey, forgot about this issue. I've come back to this embedded firmware. Currently, I've just realised I need the HAL crate as well. Anything I can do?

ByteNacked, 2021-11-15:
Hello,
All my work in  https://github.com/ByteNacked/stm32l4xx-hal/tree/stm32l4r9 branch, 
but it's a little bit dated, and I've never had time to polish it to PR.

It has barely enough peripherals for my device, though worked quite well for me.

shymega, 2021-11-15:
I'm rebasing against master. Everything applies cleanly except for two `pins!` macros. I've attached the file that doesn't rebase cleanly. Any ideas on what to keep? I'm doing this on my fork - if you want to collab on the same repo though, we can arrange that...
[i2c.rs.txt](https://github.com/stm32-rs/stm32l4xx-hal/files/7541221/i2c.rs.txt)

ByteNacked, 2021-11-15:
Commit `dd96a55 (Add basic stm32l4r9 support)` adds only some feature-gating https://github.com/ByteNacked/stm32l4xx-hal/commit/dd96a5567a60a0649caa18da939eec273d30a69c .
So you probably should keep only feature gate attributes and/or move them in the right places.

shymega, 2021-11-15:
Sure, I was referring to the `pins!()` macro usages though - unless the commit you refer to is related?

shymega, 2021-11-15:
oops. nvm.

shymega, 2021-11-15:
Hm. I'm not sure what I'm doing with these particular regions, and I need to familiarise myself with STM32-rs...I don't have a STM32L4R9, so I don't want to commit broken code.

shymega, 2021-11-25:
OK, so there are actually three files here that don't rebase cleanly against `master`:

- `src/i2c.rs`
- `src/gpio.rs`
- `src/serial.rs`

As I don't have an L4R9, I am unable to resolve these issues without hands-on knowledge, or other information, so I'm going to have to leave the rebasing for the moment. If anyone else with an STM32L4R9 can rebase against master and resolve these issues, that'd be great - we could then start adding peripherals, and I can get the HAL for this chip in this phone working firmware-side. :smile:

jspngh, 2021-11-29:
@shymega, you can also have a look at #270.

korken89, 2022-03-02:
Cool, closing

closed 2022-03-02 by korken89