stm32-rs/stm32l4xx-hal #219: Race conditions

FrozenDroid, 2021-04-26:
Hi. We've been trying to get our code to run with opt-levels above 1, but we've been having some issues.
Turns out there's some issues where code has been written specifically to work with opt-level 1 or less.
One example is https://github.com/stm32-rs/stm32l4xx-hal/blob/master/src/timer.rs#L60
Here, if you use opt-level 2 for example, this line will be executed before the update event flag is event set. Thus, the interrupt will be constantly pending because the clearing happens too soon.
I will see if I can whip up a solution, but if anyone else wants to have their hand at fixing it, that would be great. I'll let you know when I start working on a solution.

korken89, 2021-05-30:
Thanks for reporting!

Seems odd, the register access has volatile semantics so they should not reorder.
Have you had a look at the generated assembly?

FrozenDroid, 2021-05-30:
Volatile access is not relevant here. Loading the prescaler value actually just takes a while, and you're not waiting till the update flag is actually set. So you clear it even though it's not set.

korken89, 2021-06-01:
Ah, I see.
You seem to understand how the timer works in this case, could you provide a patch for this issue?