stm32-rs/stm32l4xx-hal #223: LPUART implementation in serial.rs

rgw3d, 2021-04-30:
I'd like to be able to use the lpuart capability on my STM32L496, and noticed that it wasn't already implemented! 

I might make an effort to do this myself, but was curious if someone has thought about this already and had insight as to why it wasn't implemented already.

romixlab, 2021-05-06:
Needed the same thing, came up with this, not yet tested though: https://github.com/romixlab/stm32l4xx-hal/commit/1cf33325a04bf5c40eb4cb140723e6a1605bc6c6

romixlab, 2021-05-07:
This works: https://github.com/romixlab/stm32l4xx-hal/commit/3a1b7be9c656dedcbdd02c85ddd50d927010a782

korken89, 2021-05-31:
The answer to not being implemented yet is simple, none of the maintainers has needed it yet :)
We would welcome a PR adding the LPUART.

jglauche, 2021-06-06:
> This works: [romixlab@3a1b7be](https://github.com/romixlab/stm32l4xx-hal/commit/3a1b7be9c656dedcbdd02c85ddd50d927010a782)

would you like to make this a PR? I'm in need of the same thing. I don't want to work on it too if you plan to do this soonish.

rgw3d, 2021-06-06:
I have not been able to work on this at all

romixlab, 2021-06-23:
Finally got some time to work on this, PR is open.