stm32-rs/stm32l4xx-hal #227: ADC/linker problem - won't compile (STM32L432KC)

nebelgrau77, 2021-05-20:
I'm trying to use the ADC example with an STM32L432KC. If I specify version 0.6.0 of the crate, it can't find ADC1, ADC_COMMON or rcc.ccipr. I used git = "git = "https://github.com/stm32-rs/stm32l4xx-hal" instead and it seemed to compile OK, but then it won't:

 note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" _(omitted)_
  = note: rust-lld: error: undefined symbol: ADC1_2
          >>> referenced by stm32l4.arnfbrxu-cgu.3
          >>>               stm32l4-3c82b24cdd335e86.stm32l4.arnfbrxu-cgu.3.rcgu.o:(__INTERRUPTS) in archive 

I believe I have everything necessary installed, I actually have some firmware already working on the same board, I just wanted to add ADC. Is it something I'm doing wrong, or an actual issue with the crate?

Thanks!

korken89, 2021-05-30:
Seems like an error in the PAC with the fields and interrupt definition.
Could you have a look between the PAC and the reference manual?

nebelgrau77, 2021-05-31:
I don't think I'm versed enough in those details, but I see that in [PAC ](https://github.com/stm32-rs/stm32-rs/blob/master/devices/stm32l4x2.yaml) there are the following lines:

```json
ADC1:  
    _modify:    
        _interrupts:      
            ADC1:        
                name: ADC1_2
```
and ADC1_2 also appears in the NVIC vector table in the [RM](https://www.st.com/resource/en/reference_manual/dm00151940-stm32l41xxx42xxx43xxx44xxx45xxx46xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) as ADC1 and ADC2 global interrupt (page 322). 
I don't know if it's any help to understand the problem, though.

korken89, 2021-06-01:
Hmm, unclear what is the issue here.
But the error you get are consistent with an error in the PAC.

Could you open an issue in `stm32-rs/stm32-rs` and link it here?

nebelgrau77, 2021-06-01:
OK, done: https://github.com/stm32-rs/stm32-rs/issues/561