stm32-rs/stm32l4xx-hal #233: I2C NACK polling

Crzyrndm, 2021-06-02:
current master branch asserts that the buffer length for I2C::read and I2C::write is at least 1 and less than 256 (1..=255)

write: https://github.com/stm32-rs/stm32l4xx-hal/blob/master/src/i2c.rs#L254
read: https://github.com/stm32-rs/stm32l4xx-hal/blob/master/src/i2c.rs#L308

A fairly common I2C requirement is checking if an address is present/responding by just writing the address and checking for the device ACK. Note that this is a 0 length read/write and therefore the asserts linked above prevent this.

There is some [conversation in the embedded-hal](https://github.com/rust-embedded/embedded-hal/issues/181) crate about adding an explicit method for this but no real consensus or movement.

gauteh, 2022-03-02:
Assert removed in #295 . Tested on r5.

gauteh, 2022-03-05:
Fixed for write