stm32-rs/stm32l4xx-hal #255: af4 open drain output

pdgilbert, 2021-10-01:
After the  alternate function changes https://github.com/stm32-rs/stm32l4xx-hal/commit/104b568bc063c4c7fbad9e03d3ed7d9aa4e4e5e7  I am having trouble finding the correct syntax for af4 open drain output.  I can use `.into_af4_opendrain(...)` and  `.into_open_drain_output(...)`  but I get

```
    |
651 |         (i2c, led, tx)
    |          ^^^ expected struct `Output`, found struct `OpenDrain`
    |
    = note: expected struct `I2c<_, (PA9<Alternate<_, Output<OpenDrain>>>, PA10<Alternate<_, Output<OpenDrain>>>)>`
               found struct `I2c<_, (PA9<Alternate<_, OpenDrain>>, PA10<Alternate<_, OpenDrain>>)>`

```
Variations on the obvious  `.into_af4_open_drain_output(...)` do not seem to be available.

I am also having trouble
```
error[E0599]: no method named `into_af7` found for struct `PA9<Input<Floating>>` in the current scope
   --> examples/driver-examples/max30102-heart-usart.rs:685:23
    |
685 |             gpioa.pa9.into_af7(&mut gpioa.moder, &mut gpioa.afrh),
    |                       ^^^^^^^^ method not found in `PA9<Input<Floating>>`
```
Suggestions appreciated.

korken89, 2021-10-02:
@DrTobe could you have a look?
You know this change inside and out :)

DrTobe, 2021-10-02:
Yep, for sure.

For the first issue, `.into_af4_opendrain(...)` is the right method to call for you. It does what `.into_open_drain_output(...)` followed by `.into_af4(...)` did before. The issue arises because your application probably specifies somewhere that a parameter or field must be of type `PA9<Alternate<_, Output<OpenDrain>>>` which does not exist anymore. Instead, the corresponding typestate for PA9 is now called `PA9<Alternate<_, OpenDrain>>`. This is due to the fact that alternate function pins are neither input nor output (think of i2c data pins). This is now reflected correctly in the typestates of the GPIOs.

For the second issue, `.into_af4(...)` alone does not exist anymore. You always have to specify the requested output type, either push-pull or open-drain. For most cases, this is `.into_af4_pushpull(...)`, the pin will be in state `PA9<Alternate<AF4, PushPull>>` then. Like above, this change in typestate has to be reflected by your application.

If this does not resolve your issue yet, let me have a look at the context.

pdgilbert, 2021-10-03:
Thank you @DrTobe.  All my examples that were previously working are now working again. I do have one long outstanding problem with open drain input (#144). I think this may have also been fixed by the recent alternate function changes but I'm not sure. I've just been attempting it but still with no success. I'll close this and put that question in #144. (For anyone reading this thread the CI for my examples is at https://github.com/pdgilbert/rust-integration-testing/actions. Check dev-testing jobs.)

closed 2021-10-03 by pdgilbert