stm32-rs/stm32l4xx-hal #329: Start work on adapting embedded_hal 1.0

Lol3rrr, 2022-09-13:
# Reasoning
Considering that embedded_hal is moving towards a 1.0.0 release, it might make sense to slowly start migrating or prepare to migrate this project to embedded_hal 1.0.0 as well, to stay up to date.

# Proposal / Idea
Given that embedded_hal is still working on 1.0 alphas, we might want to add a seperate branch for that as well, so we might be able to more closely follow the development of embedded_hal. I would be willing to at least contribute to this as well, but I image this would be a lot of work, which is another reason as to why we should probably start thinking about it now.
However I would be curious if there are already ideas for this and what the general support for this would be here

sousandrei, 2023-01-30:
+1 on this, some newer crates/drivers are already using the alphas on embedded-hal, which conflicts heavily with the stable 0.x.x versions we use here :cry:

Alex-Muirhead, 2023-07-22:
I also support this a lot, has there been any work towards it so far?