stm32-rs/stm32l4xx-hal #340: Register read/write not writing on stm32l433

sousandrei, 2023-04-16:
Something seems to be off.

When writing on registers using the following code, the results tells the the read/write from won't work unless I go hardcore on `ptr::read` or `ptr::write`.

Maybe I'm using the wrong register, but I'm using `stm32l4::stm32l4x3::SPI2::PTR` as a reference :thinking: 

When writing with `prt::write`, it actually works on `ptr::read` but not on the `dp` read, always zero

Am I missing something?

Context: was trying to use some SPI crate and the result was very random all the time, could not pinpoint the error so started digging down

```rust
let dp = stm32::Peripherals::take().unwrap();
let mut spi2 = 0x4000_3800;

writeln!(sys, "dp read {:?}", dp.SPI2.cr1.read().bits()).unwrap();
writeln!(sys, "core read {:?}", unsafe { core::ptr::read(&spi2) }).unwrap();

dp.SPI2.cr1.modify(|_, w| w.spe().enabled());

writeln!(sys, "dp read {:?}", dp.SPI2.cr1.read().bits()).unwrap();
writeln!(sys, "core read {:?}", unsafe { core::ptr::read(&spi2) }).unwrap();

unsafe { core::ptr::write(&mut spi2, 1) }
writeln!(sys, "core write 1").unwrap();

writeln!(sys, "dp read {:?}", dp.SPI2.cr1.read().bits()).unwrap();
writeln!(sys, "core read {:?}", unsafe { core::ptr::read(&spi2) }).unwrap();
```

Resulting in
```
23:34:06.743 dp read 0
23:34:06.743 core read 1073756160

23:34:06.743 dp read 0
23:34:06.743 core read 1073756160

23:34:06.743 core write 1

23:34:06.743 dp read 0
23:34:06.743 core read 1
```

snakehand, 2023-10-21:
Seems there is something wrong with your code as the decimal value 1073756160 = 0x40003800  , so I suspect you are just reading the variable you have declared, and not the memory adress it points to.

snakehand, 2023-10-21:
Also you should make sure that the SPI2 device is actually enabled by setting SPI2EN bit in APB1ENR1 register, otherwise the memory mapped registers typically will just read 0.